tock/libtock-c #44: How to support RISC-V apps?

bradjc, 2019-06-11:
Do we want to expect libtock-c users to have the riscv compiler installed and have it be built every time?

ppannuto, 2019-06-13:
I mean, you could ask the equivalent question of should we require RISC-V users to have the ARM compiler installed and building unused Thumb binaries. There's a toolchain / fragmentation tradeoff here.

My lean is to say that adding one more package to the "install this" list of copy-pasted `apt`/`brew`/whatever isn't so painful -- this strongly assumes that installing the toolchain is effectively a no-op (which I think is true). We could certainly configure the build system to do runtime detection (e.g. guarding all of the arch-specific build rules on the result of a `if $$(which $(CC))`), but I don't think that's worth the maintenance

bradjc, 2019-10-01:
Update: no real progress on this issue. If you are here because you are looking for RISC-V support, look for the `riscv` [branch](https://github.com/tock/libtock-c/tree/riscv) which should be able to build apps for RISC-V platforms.

Actual progress is probably blocked on being able to support some sort of position-independent applications. Right now compiling apps for RISC-V is very cludgy since they have to be compiled for the address they are eventually placed at.

hudson-ayers, 2020-06-08:
In general I agree with Pat on this, but if the install process for Linux is much more complicated / distro-dependent than an apt install, I would push back. I don't think that asking all linux users to build the toolchain from source and install it is particularly reasonable, for example (which seems like what most Linux users have been doing up until now)

ppannuto, 2020-06-30:
I think this is closed by #88.

closed 2020-06-30 by ppannuto