tock/libtock-c #48: Linker script has a dead URL in the comments

JayKickliter, 2019-07-16:
Located at @ https://github.com/tock/libtock-c/blob/master/userland_generic.ld#L134

Side note: I'm not doing any Tock development at the moment, but Tock's linker scripts are so well documented that I use them as references. Thanks!

ppannuto, 2019-07-16:
Looks like they've taken that document down, from https://wiki.linaro.org/FrontPage

> The Linaro Developer Wiki
>
> Please note!
>
> The developer wiki has been archived and we are starting a fresh new one. If you have a Linaro login account and cannot find the information you were looking for on this wiki, please check the Wiki Archive.

---

Scanning around a bit, I've found dozens of other people linking to that document, but no one hosting it. It's doubly a shame, as lots of people referencing it refer to it as a great guide :(

Guess we should just delete the link unless anyone has an idea of how to recover a copy? Couldn't find it on the internet archive either :/

ppannuto, 2020-03-27:
Aha! Patience pays dividends! Linario has updated the archive of their wiki that did not have this page the last time I looked, but now https://wiki-archive.linaro.org/KenWerner/Sandbox/libunwind is alive!

For at least one archival source, here's a snapshot of the wiki page and the presentation:

[KenWerner-libunwind.pdf](https://github.com/tock/libtock-c/files/4395453/KenWerner-libunwind.pdf)
[libunwind-LDS.pdf](https://github.com/tock/libtock-c/files/4395455/libunwind-LDS.pdf)

PR #77 (2020-03-30): Update dead link in documentation
Closes #48.
commit faea55f1 (2020-03-30): Update dead link in documentation

Closes #48.
closed 2020-03-30 by brghena