tock/libtock-c #93: 'ble-uart' example panics in hail board

JOE1994, 2020-07-08:
Hello :crab: ,

`libtock-c/examples/ble-uart/README.md` mentions that the `ble-uart` example works on `imix` & `hail`,
but the program panics on my `hail` board due to **Data Access Violation** fault.

`ble-uart` was the only app installed on top of Tock when it panicked.

* debug messages from `tockloader listen`
```
Listening for serial output.
␀
Initialization complete. Entering main loop.

panicked at 'Restart threshold surpassed!', kernel/src/process.rs:532:13
        Kernel version release-1.5-696-gf2b110dd

---| No debug queue found. You can set it with the DebugQueue component.

---| Fault Status |---
Data Access Violation:              true
Forced Hard Fault:                  true
Faulting Memory Address:            0x20005000
Fault Status Register (CFSR):       0x00000082
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: ble-uart   -   [StoppedFaulted]
 Events Queued: 0   Syscall Count: 3   Dropped Callback Count: 0
 Restart Count: 5
 Last Syscall: Some(MEMOP { operand: 11, arg0: 536896160 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant        1036 |   1036          
  0x20005BF4 ┼───────────────────────────────────────────
             │ Unused
  0x20004C00 ┼───────────────────────────────────────────
             │ ▲ Heap            ? |      ?               S
  ?????????? ┼─────────────────────────────────────────── R
             │ Data              ? |      ?               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack          40 |   2048          
  0x200047D8 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00040000 ┬─────────────────────────────────────────── F
             │ App Flash     65492                        L
  0x0003002C ┼─────────────────────────────────────────── A
             │ Protected        44                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x20004B60    R6 : 0x0003002C
  R1 : 0x00000000    R7 : 0x20004000
  R2 : 0x200062A0    R8 : 0x00000000
  R3 : 0x20005000    R10: 0x00000000
  R4 : 0x0003002C    R11: 0x00000000
  R5 : 0x20004000    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047D8 (Process Stack Pointer)
  LR : 0x00030429
  PC : 0x000367E8
 YPC : 0x00030088

 APSR: N 1 Z 0 C 0 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Cortex-M MPU
  Region 0: [0x20004000:0x20006000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20004000:0x20004400], Enabled
    Sub-region 1: [0x20004400:0x20004800], Enabled
    Sub-region 2: [0x20004800:0x20004C00], Enabled
    Sub-region 3: [0x20004C00:0x20005000], Enabled
    Sub-region 4: [0x20005000:0x20005400], Disabled
    Sub-region 5: [0x20005400:0x20005800], Disabled
    Sub-region 6: [0x20005800:0x20005C00], Disabled
    Sub-region 7: [0x20005C00:0x20006000], Disabled
  Region 1: [0x00030000:0x00040000], length: 65536 bytes; ReadOnly (0x6)
    Sub-region 0: [0x00030000:0x00032000], Enabled
    Sub-region 1: [0x00032000:0x00034000], Enabled
    Sub-region 2: [0x00034000:0x00036000], Enabled
    Sub-region 3: [0x00036000:0x00038000], Enabled
    Sub-region 4: [0x00038000:0x0003A000], Enabled
    Sub-region 5: [0x0003A000:0x0003C000], Enabled
    Sub-region 6: [0x0003C000:0x0003E000], Enabled
    Sub-region 7: [0x0003E000:0x00040000], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x30055`
in the app's folder and open the .lst file.
```

I followed the direction to debug by running `make debug RAM_START=0x20004000 FLASH_INIT=0x30055` in the app's
folder and got a `.lst` file consisting of 708205 lines, with which I'm not sure what to do :crying_cat_face:

bradjc, 2020-07-08:
Hi thanks for reminding me about this issue. The issue is that elf2tab
isn't calculating the correct amount of RAM for the app after changes to
how the stack is specified. If you install elf2tab from the repo that will
fix it (release pending). Otherwise `make APP_HEAP_SIZE=4096` should work
as well.

JOE1994, 2020-07-09:
> Hi thanks for reminding me about this issue. The issue is that elf2tab isn't calculating the correct amount of RAM for the app after changes to how the stack is specified. If you install elf2tab [...]

Thank you so much for your help! :superhero:  After installing `elf2tab` from the repo
(`cargo install --git https://github.com/tock/elf2tab.git`) and rebuilding the example,
now the example runs successfully on my hail board!

closed 2020-12-13 by hudson-ayers