tock/libtock-c #95: printing formatted string doesn't work (e.g.) printf("%f\n", 123.456)

JOE1994, 2020-07-12:
Hello :crab: ,

Below is a slightly modified version of the [`cxx_hello`](https://github.com/tock/libtock-c/blob/master/examples/cxx_hello/main.cc) example.
The example builds without errors, **123.456** is not printed to the console output.
Newline character in `"%f\n"` is printed out to console,
but the formatted string(formatted floating point number) is not printed to the console.

The example was tested on the `Hail` board.

* Slightly modified 
```rust
int main() {
  printf("Hello World\n");   // Newly added line
  printf("%f\n", 123.456);     // Newly added line

  volatile uint8_t test_branch = 0;

  D1 d1class;
  D2 d2class;
  Base *pClass;
  while (1) {
    if (test_branch % 2) {
      pClass = &d1class;
      pClass->function1();
    } else {printf
      pClass = &d2class;
      pClass->function1();
    }
    test_branch++;

    delay_ms(1000);
  }

  return 0;
}
```

* Program output (float **123.456** is not printed to console)
```
Initialization complete. Entering main loop.
Hello World

D2 says hello
D1 says hello
D2 says hello
D1 says hello
D2 says hello
```

Thank you for checking out this issue :+1:

hudson-ayers, 2020-07-13:
The C compiler we use for libtock-c disables float formatting by default to reduce code size. If you need to print formatted floats, you can add the following line to the Makefile of your app:

`override CFLAGS += -Wl,-u,_printf_float`

The odd syntax is because the libtock-c Makefile does not pass flags directly to ld, so you have to use the -Wl flag to tell gcc to pass a comma-seperated list of arguments to the linker.

JOE1994, 2020-07-14:
> The C compiler we use for libtock-c disables float formatting by default to reduce code size. If you need to print formatted floats, you can add the following line to the Makefile of your app:
> 
> `override CFLAGS += -Wl,-u,_printf_float`
> 
> The odd syntax is because the libtock-c Makefile does not pass flags directly to ld, so you have to use the -Wl flag to tell gcc to pass the comma-seperated list of arguments to the linker.

Oh, I see.. Thank you for your help :smile_cat:

closed 2020-07-14 by JOE1994