tock/libtock-c #108: Building applications inside a docker container fails

skyplane23, 2020-09-28:
I & @alexandruradovici are trying to build a simple application inside a docker container but it fails. All paths seem to be fine but we keep receiving "fatal error: stdio.h: No such file or directory" and we can't figure out why. Can you please assist us?

```bash
root@7b4ba3df2e65:~/libtock-c/examples/c_hello# make V=1

**************************************************
TOCK USERLAND BUILD SYSTEM -- VERBOSE BUILD
**************************************************
Config:
CC=-gcc
LAYOUT=../../userland_generic.ld
MAKEFLAGS= -r -R
PACKAGE_NAME=c_hello
TOCK_ARCHS=cortex-m0 cortex-m3 cortex-m4
TOCK_TARGETS=cortex-m0 cortex-m3 cortex-m4
TOCK_USERLAND_BASE_DIR=../..
TOOLCHAIN=
**************************************************

arm-none-eabi-gcc -std=gnu11 -Wbad-function-cast  -Wjump-misses-init  -Wmissing-prototypes  -Wnested-externs  -Wold-style-definition  -frecord-gcc-switches -gdwarf-2 -Os -fdata-sections -ffunction-sections -fstack-usage -Wstack-usage=2048 -Wall -Wextra -Wl,--warn-common -Wl,--gc-sections -include ../../support/warning_header.h -Wdate-time  -Wfloat-equal  -Wformat-nonliteral  -Wformat-security  -Wformat-y2k  -Winit-self  -Wlogical-op  -Wmissing-declarations  -Wmissing-field-initializers  -Wmissing-format-attribute  -Wmissing-noreturn  -Wmultichar  -Wpointer-arith  -Wredundant-decls  -Wshadow  -Wtrampolines  -Wunused-macros  -Wunused-parameter  -Wwrite-strings  -I../../libtock/ -I../../libtock//include -I../../libtock//../include -I../../libtock/internal/ -I../../libtock/internal//include -I../../libtock/internal//../include -Wl,--emit-relocs -fPIC -mthumb -mfloat-abi=soft -msingle-pic-base -mpic-register=r9 -mno-pic-data-is-text-relative -mcpu=cortex-m0 -march=armv6s-m -MF"../../libtock/build/cortex-m0/alarm_internal.d" -MG -MM -MP -MT"../../libtock/build/cortex-m0/alarm_internal.d@" -MT"../../libtock/build/cortex-m0/alarm_internal.o" "../../libtock/internal/alarm_internal.c"
arm-none-eabi-gcc -std=gnu11 -Wbad-function-cast  -Wjump-misses-init  -Wmissing-prototypes  -Wnested-externs  -Wold-style-definition  -frecord-gcc-switches -gdwarf-2 -Os -fdata-sections -ffunction-sections -fstack-usage -Wstack-usage=2048 -Wall -Wextra -Wl,--warn-common -Wl,--gc-sections -include ../../support/warning_header.h -Wdate-time  -Wfloat-equal  -Wformat-nonliteral  -Wformat-security  -Wformat-y2k  -Winit-self  -Wlogical-op  -Wmissing-declarations  -Wmissing-field-initializers  -Wmissing-format-attribute  -Wmissing-noreturn  -Wmultichar  -Wpointer-arith  -Wredundant-decls  -Wshadow  -Wtrampolines  -Wunused-macros  -Wunused-parameter  -Wwrite-strings  -I../../libtock/ -I../../libtock//include -I../../libtock//../include -I../../libtock/internal/ -I../../libtock/internal//include -I../../libtock/internal//../include -Wl,--emit-relocs -fPIC -mthumb -mfloat-abi=soft -msingle-pic-base -mpic-register=r9 -mno-pic-data-is-text-relative -mcpu=cortex-m0 -march=armv6s-m -c -o ../../libtock/build/cortex-m0/alarm_internal.o ../../libtock/internal/alarm_internal.c
In file included from <command-line>:
./../../support/warning_header.h:6:10: fatal error: stdio.h: No such file or directory
    6 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.
make: *** [../../TockLibrary.mk:149: ../../libtock/build/cortex-m0/alarm_internal.o] Error 1
```

Content of the docker file:
```docker
FROM ubuntu:20.04

WORKDIR /wyliodrin
ENV HOME=/wyliodrin

# install python3 and build tools
RUN apt-get update && apt-get install -y --no-install-recommends python3 python3-pip build-essential curl openocd git gcc-arm-none-eabi

# install nodejs
RUN curl https://nodejs.org/dist/v12.18.4/node-v12.18.4-linux-x64.tar.xz -sSf | tar xvJ &&\
    cd node* && cp -R * /usr && cd .. && rm -rf node* && node -v

# install rust
RUN curl https://sh.rustup.rs -sSf | sh -s -- -y
RUN pip3 install --upgrade tockloader

# install tock os
RUN git clone https://github.com/tock/tock.git
RUN git clone https://github.com/tock/libtock-c.git

# install studio-supervisor
COPY . /wyliodrin/studio-supervisor
RUN cd studio-supervisor && npm install 

ENV TOCK_DIR=/wyliodrin/tock
ENV PATH="${PATH}:/${HOME}/.cargo/bin"

CMD cd /wyliodrin/studio-supervisor/source && node startup.js tockos
```

ppannuto, 2020-09-28:
Docker provides a very restricted system, which means you will likely need to manually install (possibly many) packages that would normally already be present in a traditional environment. For `stdio.h` specifically, I believe you need `sudo apt-get install libc6-dev`.

skyplane23, 2020-10-01:
Thank you for your answer! I've installed ```libc6-dev```, but I still receive the same error. Also, I wrote a simple C program and compiles without any problems. Do you have other suggestions that I may try?

```C
#include <stdio.h>

int main(void) {
    printf("test\n");

    return 0;
}
```

ppannuto, 2020-10-01:
Huh. Interesting.. that's particularly odd. I don't know what's going on here sadly, I'd try:

 1. Copy the whole compile invocation with all the flags and such from above on that mini-program -- if that fails, iterate until you find where the problem is between your MWE and that; if that works..
 2. Add the `warning_header.h` include to your MWE and see if that's causing problems?

Hope that helps...

skyplane23, 2020-10-02:
In the end, we figure it out. Because in the container we installed ```gcc-arm-none-eabi``` with the flag ```--no-install-recommends``` 
something was left uninstalled and we had to install it manually afterward. The package that missed was ```libnewlib-arm-none-eabi```. We found this answer that helped us: https://askubuntu.com/questions/502815/should-arm-none-eabi-gcc-include-a-stdio-h.

closed 2020-10-02 by skyplane23