tock/libtock-c #114: Toolchain configuration

towoe, 2020-10-19:
Is there a reason why the `TOOLCHAIN_*` variables are set with the `override` prefix in `Configuration.mk`?

The values for `TOOLCHAIN_rv32im[a]c` are currently `riscv64-unknown-elf`.
Some packages (e.g. Arch, which is also referenced in the README.md) have binaries which have a prefix `riscv64-elf`.
It should also be possible to compile with `riscv32-unknown-elf-gcc`.
Not using `override` would help to configure which compiler is used locally.

The configuration dump of `Configuration.mk` uses the variable `TOOLCHAIN` but as far as I can see it is never used, to this output is a bit confusing.

ppannuto, 2020-10-19:
`override` is necessary for the FLAGS variables and several other configuration ones -- it was probably a bit of an overzealous pass of adding override to everything at some point in the past. I don't think there's a specific need for TOOLCHAIN to be override.

Ideally, we'd probably do a pass and leave override for everything with a `+=` directive and drop it for all the `:=` 's

towoe, 2020-10-20:
Thanks for the quick reply @ppannuto and explanation!
I removed the keyword in the PR, if there are more changes you would like me to add to the PR feel free to comment.

commit 6363dd8b (2020-10-20): Allow toolchains to bet set

Remove keyword `override` so the variables are changeable with make
arguments.

Fixes #114
commit 35570a30 (2020-10-20): Merge #115

115: Allow toolchains to bet set r=ppannuto a=towoe

Remove keyword `override` so the variables are changeable with make
arguments.

Fixes #114
PR #115 (2020-10-20): Allow toolchains to bet set
Remove keyword `override` so the variables are changeable with make
arguments.

Fixes #114
closed 2020-10-20 by bors[bot]