tock/libtock-c #117: Required elf2tab version

towoe, 2020-10-30:
Are there any requirements on which `elf2tab` version must be used in _libtock-c_?

I must have built something with _libtock-rs_ first, because I have version 0.4.0 installed as specified in the [libtock-rs Makefile](https://github.com/tock/libtock-rs/blob/593df520471e29fcd26a38071b5010cdb79a5998/Makefile#L46)

The test in the configuration will succeed and not pull the current version:
https://github.com/tock/libtock-c/blob/53af0d2660f9d09338746df1896aa271c8b602b3/Configuration.mk#L75

And it seems like [shell.nix](https://github.com/tock/libtock-c/blob/53af0d2660f9d09338746df1896aa271c8b602b3/shell.nix#L37) will use version 0.3.0

lschuermann, 2020-10-30:
> And it seems like [shell.nix](https://github.com/tock/libtock-c/blob/53af0d2660f9d09338746df1896aa271c8b602b3/shell.nix#L37) will use version 0.3.0

That is correct. I cannot make any statements regarding the minimum required version, however for RISC-V I've found that v0.6.0 will work. Please see my related [PR regarding a Cargo.lock update](https://github.com/tock/elf2tab/pull/26). As soon as that's merged, I'll make a PR to update `shell.nix` to the current release.

Pinging @bradjc as this is likely an action item for him.

towoe, 2020-10-30:
Thanks for the feedback @lschuermann. I also had some problems with elf2tab 0.4.0 and RISC-V.

Not sure if https://github.com/tock/libtock-c/pull/109 requires a specific elf2tab version, but I had trouble getting the headers to the correct position with elf2tab v0.4.0. Maybe @alistair23  has some insight on the requirements here.

alistair23, 2020-10-30:
I just use the latest version (0.6.0) and don't have any issues.

towoe, 2020-10-31:
Yes the `--protected-region-size` argument helped. This is what is used in [libtock-rs](https://github.com/tock/libtock-rs/blob/master/tools/flash.sh#L73).
Or using elf2tab v0.6.0 could also resolve the issue with the header, that's why I wanted to know what the required version is.
I guess either one should be used and documented here as well because the default settings might not produce the correct output.

towoe, 2020-10-31:
Okay, the documentation actually says something about the version: [up-to-date](https://github.com/tock/libtock-c/blame/master/README.md#L139).

lschuermann, 2020-10-31:
> Okay, the documentation actually says something about the version: [up-to-date](https://github.com/tock/libtock-c/blame/master/README.md#L139).

Great. I'd actually prefer to keep it that way. Introducing yet another number to bump would probably cause that to get outdated quickly. Great to hear that `v0.6.0` solves the issues for you.

towoe, 2020-11-03:
I think it should be documented what version is required. Maybe that should be done centrally so bumping dependencies is clearly defined.
Especially with a check and installation by the build system the user will not suspect that an incompatible version is installed.
And with a valid looking generated output it requires deeper knowledge to find the problem.

ppannuto, 2020-11-14:
I'm not sure there's a clean way to mark this in a single central place -- with libtock-rs now being up to date, hopefully this won't be an issue for folks moving forward, so I'm going to close this out for now.

closed 2020-11-14 by ppannuto