tock/libtock-c #243: arm applications fault

alexandruradovici, 2021-08-18:
I tried running the tests with the new version of `newlib` using a Micro:bit v2 and a Arduino Nano RP2040 Connect. Applications seem to fault on both platforms. The fault seems to be in `newlib` in `memcpy`.

Arduino Nano RP2040 Connect
```
RP2040 Revisi
panicked at 'Process blink had a fault', kernel/src/process_standard.rs:323:17
	Kernel version release-2.0-rc2-68-g2cad28a17

---| Debug buffer not empty. Flushing. May repeat some of last message(s):
Initialization complete. Enter main loop

---| No debug queue found. You can set it with the DebugQueue component.

---| Cortex-M Fault Status |---
No Cortex-M faults detected.

---| App Status |---
𝐀𝐩𝐩: blink   -   [Faulted]
 Events Queued: 0   Syscall Count: 3   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Some(Memop { operand: 11, arg0: 536889856 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant        1056 |   1056          
  0x20005BE0 ┼───────────────────────────────────────────
             │ Unused
  0x20004A00 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4576               S
  0x20004A00 ┼─────────────────────────────────────────── R
             │ Data            512 |    512               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        2048 |   2048          
  0x20004000 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x10020800 ┬─────────────────────────────────────────── F
             │ App Flash      1996                        L
  0x10020034 ┼─────────────────────────────────────────── A
             │ Protected        52                        S
  0x10020000 ┴─────────────────────────────────────────── H

  R0 : 0x2000482D    R6 : 0x00000000
  R1 : 0x10020738    R7 : 0x00000000
  R2 : 0x00000070    R8 : 0x10020034
  R3 : 0x00000704    R10: 0x1002070C
  R4 : 0x00000000    R11: 0x80000000
  R5 : 0x20004800    R12: 0x89745803
  R9 : 0x20004000 (Static Base Register)
  SP : 0x200047C8 (Process Stack Pointer)
  LR : 0x10020101
  PC : 0x10020A56
 YPC : 0x1002008A

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Total number of grant regions defined: 6
  Grant  0 : --          Grant  2 : --          Grant  4 : --        
  Grant  1 : --          Grant  3 : --          Grant  5 : --        

 Cortex-M MPU
  Region 0: [0x20004000:0x20006000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20004000:0x20004400], Enabled
    Sub-region 1: [0x20004400:0x20004800], Enabled
    Sub-region 2: [0x20004800:0x20004C00], Enabled
    Sub-region 3: [0x20004C00:0x20005000], Disabled
    Sub-region 4: [0x20005000:0x20005400], Disabled
    Sub-region 5: [0x20005400:0x20005800], Disabled
    Sub-region 6: [0x20005800:0x20005C00], Disabled
    Sub-region 7: [0x20005C00:0x20006000], Disabled
  Region 1: [0x10020000:0x10020800], length: 2048 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x10020000:0x10020100], Enabled
    Sub-region 1: [0x10020100:0x10020200], Enabled
    Sub-region 2: [0x10020200:0x10020300], Enabled
    Sub-region 3: [0x10020300:0x10020400], Enabled
    Sub-region 4: [0x10020400:0x10020500], Enabled
    Sub-region 5: [0x10020500:0x10020600], Enabled
    Sub-region 6: [0x10020600:0x10020700], Enabled
    Sub-region 7: [0x10020700:0x10020800], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x1002005d`
in the app's folder and open the .lst file.
```

Micro:bit v2

```
Initialization complete. Entering main loop.

panicked at 'Process blink had a fault', kernel/src/process_standard.rs:323:17
	Kernel version release-2.0-rc2-68-g8ccabf224

---| No debug queue found. You can set it with the DebugQueue component.

---| Cortex-M Fault Status |---
Instruction Access Violation:       true
Undefined Instruction Usage Fault:  true
Forced Hard Fault:                  true
Fault Status Register (CFSR):       0x00010001
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
𝐀𝐩𝐩: blink   -   [Faulted]
 Events Queued: 0   Syscall Count: 3   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Some(Memop { operand: 11, arg0: 536889856 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant        1120 |   1120          
  0x20005BA0 ┼───────────────────────────────────────────
             │ Unused
  0x20004A00 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4512               S
  0x20004A00 ┼─────────────────────────────────────────── R
             │ Data            512 |    512               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        2048 |   2048          
  0x20004000 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00040800 ┬─────────────────────────────────────────── F
             │ App Flash      1996                        L
  0x00040034 ┼─────────────────────────────────────────── A
             │ Protected        52                        S
  0x00040000 ┴─────────────────────────────────────────── H

  R0 : 0x2000482D    R6 : 0x00040034
  R1 : 0x00040654    R7 : 0x20004000
  R2 : 0x00000070    R8 : 0x00000000
  R3 : 0x0000002C    R10: 0x00000000
  R4 : 0x00040034    R11: 0x00000000
  R5 : 0x00000000    R12: 0x7B0FFFE7
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047D8 (Process Stack Pointer)
  LR : 0x00040103
  PC : 0x00040A16
 YPC : 0x0004008E

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Total number of grant regions defined: 14
  Grant  0 : --          Grant  5 : --          Grant 10 : --        
  Grant  1 : --          Grant  6 : --          Grant 11 : --        
  Grant  2 : --          Grant  7 : --          Grant 12 : --        
  Grant  3 : --          Grant  8 : --          Grant 13 : --        
  Grant  4 : --          Grant  9 : --        

 Cortex-M MPU
  Region 0: [0x20004000:0x20006000], length: 8192 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20004000:0x20004400], Enabled
    Sub-region 1: [0x20004400:0x20004800], Enabled
    Sub-region 2: [0x20004800:0x20004C00], Enabled
    Sub-region 3: [0x20004C00:0x20005000], Disabled
    Sub-region 4: [0x20005000:0x20005400], Disabled
    Sub-region 5: [0x20005400:0x20005800], Disabled
    Sub-region 6: [0x20005800:0x20005C00], Disabled
    Sub-region 7: [0x20005C00:0x20006000], Disabled
  Region 1: [0x00040000:0x00040800], length: 2048 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x00040000:0x00040100], Enabled
    Sub-region 1: [0x00040100:0x00040200], Enabled
    Sub-region 2: [0x00040200:0x00040300], Enabled
    Sub-region 3: [0x00040300:0x00040400], Enabled
    Sub-region 4: [0x00040400:0x00040500], Enabled
    Sub-region 5: [0x00040500:0x00040600], Enabled
    Sub-region 6: [0x00040600:0x00040700], Enabled
    Sub-region 7: [0x00040700:0x00040800], Enabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x4005d`
in the app's folder and open the .lst file.
```

ppannuto, 2021-08-19:
Hmm.. I tested the new newlib on arm when I first built it several months ago, but I just pulled in those patches for the most recent PR. I'm on the road again (for the last time though! #HolyWeddingsBatman) until Monday, so won't be able to do anything useful until then. The first thing I would probably try is just running the `make rebuild` to force it to build a fresh copy of the toolchain as the patched versions are admittedly older.

alexandruradovici, 2021-08-19:
Do you mean running `build-arm.sh` in newlib's folder?

alexandruradovici, 2021-08-19:
> Hmm.. I tested the new newlib on arm when I first built it several months ago, but I just pulled in those patches for the most recent PR. I'm on the road again (for the last time though! [...]

Rebuilding solves it. But there is a downside:
- armv6 works for both Arduino Nano RP2040 Connect (Raspberry Pi Pico) and Micro:bit
- armv7 works only for Micro:bit (as expected)

This means that it might be a good idea to shop both versions of the binaries.

ppannuto, 2021-08-19:
hmm.. armv6 should work for all, v7 is a superset of v6; using v6 for Micro:bit crashes (or works, just less performant?)

we certainly can ship both though; just need to tweak the makefiles a bit to pull libraries on a per-cortex platform now

hudson-ayers, 2021-08-20:
If v6 works for all platforms, it sounds like it might be a good idea to update the v6 binary for now and then update the Makefiles to support both in a follow up PR? This is a bad time for libtock-c not to work since people are doing release testing.

alexandruradovici, 2021-08-20:
I can try to modify the makefiles to use separate versions.

bradjc, 2021-08-20:
It is straightforward to have cortex-m0 use different libc. The trickier part is getting a v6 version to build.

alexandruradovici, 2021-08-20:
The build script worked out of the box, just that, even tought it is called `build-arm`, it tried to build the riscv version also.

hudson-ayers, 2021-08-20:
if you pass the directory you want to build as a command line argument it will build only that directory

lschuermann, 2021-08-20:
#244 should have solved this, right? @alexandruradovici can you verify that this issue is indeed solved on these platforms?

alexandruradovici, 2021-08-22:
It works now.

closed 2021-08-22 by alexandruradovici