tock/libtock-c #262: Build Failures: ./courses/2018-11-SenSys/sensys_udp_rx  after  ./build_all.sh

nicolast0604, 2021-09-07:
Hi
I encountered 
Build Failures: 
./courses/2018-11-SenSys/sensys_udp_rx

after
cd examples
./build_all.sh

Please help, thanks a lot.

hudson-ayers, 2021-09-08:
Hi, could you post some more of the output log? build_all.sh completes successfully on my machine (and in our CI). What OS are you developing on?

nicolast0604, 2021-09-09:
I am using Ubuntu 18.04.4 with gcc-risc64-unknown by this way,
$ wget http://cs.virginia.edu/~bjc8c/archive/gcc-riscv64-unknown-elf-8.3.0-ubuntu.zip
$ unzip gcc-riscv64-unknown-elf-8.3.0-ubuntu.zip
# add gcc-riscv64-unknown-elf-8.3.0-ubuntu/bin to your `$PATH` variable.

ppannuto, 2021-09-09:
Could you post the actual build failure here as well? Try building just that app (go into that app  directory and run `make`)

nicolast0604, 2021-09-10:
Build Failures:
./courses/2018-11-SenSys/sensys_udp_rx

build  main.c  Makefile  README.md  server
nicolast0604@nicolast0604-VirtualBox:~/tock/libtock-c/examples/courses/2018-11-SenSys/sensys_udp_rx$ make
  CC        main.c
main.c: In function 'callback':
main.c:49:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ipv6_addr_t sender_addr = *((ipv6_addr_t*)BUF_BIND_CFG);
   ^~~~~~~~~~~
  LD        build/cortex-m0/cortex-m0.elf
  CC        main.c
main.c: In function 'callback':
main.c:49:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ipv6_addr_t sender_addr = *((ipv6_addr_t*)BUF_BIND_CFG);
   ^~~~~~~~~~~
  LD        build/cortex-m3/cortex-m3.elf
  CC        main.c
main.c: In function 'callback':
main.c:49:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ipv6_addr_t sender_addr = *((ipv6_addr_t*)BUF_BIND_CFG);
   ^~~~~~~~~~~
  LD        build/cortex-m4/cortex-m4.elf
  CC        main.c
main.c: In function 'callback':
main.c:49:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ipv6_addr_t sender_addr = *((ipv6_addr_t*)BUF_BIND_CFG);
   ^~~~~~~~~~~
  LD        build/cortex-m7/cortex-m7.elf
Application size report for target cortex-m0:
   text	   data	    bss	    dec	    hex	filename
   7816	    344	   2732	  10892	   2a8c	build/cortex-m0/cortex-m0.elf
Application size report for target cortex-m3:
   text	   data	    bss	    dec	    hex	filename
   7280	    344	   2732	  10356	   2874	build/cortex-m3/cortex-m3.elf
Application size report for target cortex-m4:
   text	   data	    bss	    dec	    hex	filename
   7280	    344	   2732	  10356	   2874	build/cortex-m4/cortex-m4.elf
Application size report for target cortex-m7:
   text	   data	    bss	    dec	    hex	filename
   7280	    344	   2732	  10356	   2874	build/cortex-m7/cortex-m7.elf

hudson-ayers, 2021-09-10:
Hmm, you must have a newer version of arm-none-eabi-gcc than I do (or our CI uses) and that is surfacing some new warnings. If you do not need to use this particular app, feel free to just navigate to the directory of the example you want to build and run `make`, that way this warnings in this app will not give you issues.

Hopefully someone will submit a fix for these warnings soon.

ppannuto, 2021-09-10:
I just updated my compiler:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.07) 10.3.1 20210621 (release)
```
and am not seeing this issue.

What compiler version are you using (run the command above)? Also, are you on an up-to-date checkout (output of `git describe --always`)?

nicolast0604, 2021-09-13:
# version of arm-none-eabi-gcc
nicolast0604@nicolast0604-VirtualBox:~/tock/libtock-c$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

#git for checkout
nicolast0604@nicolast0604-VirtualBox:~/tock/libtock-c$ git describe --always
release-1.3-684-g1e16560

ppannuto, 2021-09-13:
Aha! Other way around it seems -- that's a super old compiler, from 2017 it looks like. Try updating to a newer compiler and see if that doesn't make the issue go away :)

nicolast0604, 2021-09-14:
Great thanks! after update the compiler, it works well.

closed 2021-09-14 by lschuermann