tock/libtock-c #264: Build for RISC-V by Default

bradjc, 2021-09-13:
Currently (as of Sep 2021 at least), to build libtock-c apps for RISC-V boards requires calling `make RISCV=1`. The reason for this is that it can be hard to get a working RISC-V toolchain easily on every host platform, so it is safer to make building RISC-V apps optional.

However, only building apps for some platforms is against the spirit of building Tock apps (and the .tab format as well), so at the point where it is easy to get a working RISC-V toolchain on every common platform we should remove the `RISCV=1` flag entirely and building RISC-V apps by default.

bradjc, 2022-09-02:
I guess if fedora could get on board we could probably make the switch? Ubuntu has 22.04 which is an LTS release and newer than 21.10 where toolchain support was added.

lschuermann, 2022-09-02:
I know it's a _very_ niche platform but, for what it's work, Nix(OS) shouldn't cause any troubles here. Same with Arch Linux.

I think the issues around RISC-V are pivoting from obtainability to useability. It's no longer very hard to get a toolchain which is technically useable for the target platforms, but getting the exact right one can be tricky. In theory all RISC-V toolchains should be compatible with all architecture variants, but in practice compiler intrinsics (e.g., `libgcc`) and `libc`s are causing issues here. Also, naming conventions deviate between architectures, which can also cause confusion. If we build RISC-V by default, we might need to document installation procedures for even more platforms than we do currently.

bradjc, 2022-09-12:
> If we build RISC-V by default, we might need to document installation procedures for even more platforms than we do currently.

I have no issue with that. I do think we have to draw the line at some reasonable point, and switch to "if you cannot get a risc-v toolchain then run `make NORISCV=1`".

bradjc, 2024-02-09:
Added in #353

closed 2024-02-09 by bradjc