tock/libtock-c #267: does libtock-c support risc-v?

nicolast0604, 2021-09-17:
I ask this question is because I can't find the tbf with risc-v and only see arm cortex m0,m3,m4,m7

Is it possible to have tbf with risc-v in opentitan nexys video?

lschuermann, 2021-09-17:
Yes, we do support RISC-V! 🙂  Try building with `make RISCV=1`.

nicolast0604, 2021-09-17:
Hi
I put the commnad
$make RISCV=1
and then
$cd examples
$./build_all.sh

However, I can't find any tbf releated to riscv, do I miss anything?

lschuermann, 2021-09-17:
Okay, so there's a few things happening there: libtock-c applications are generally compiled through a Makefile-based infrastructure with an entrypoint of `AppMakefile.mk`:

https://github.com/tock/libtock-c/blob/5350f80f73949121cb98fc1c215455d4cf22b524/examples/c_hello/Makefile#L1-L11

This `AppMakefile.mk` loads the `Configuration.mk` where different options are set (some of them depending on environment variables). These options determine what gets compiled when you run `make` from within one of the example applications folder. The interesting part for you is the `TOCK_TARGETS` definition, which defines for what targets Tock applications are built. This is set depending on the environment variable `RISCV`:

https://github.com/tock/libtock-c/blob/5350f80f73949121cb98fc1c215455d4cf22b524/Configuration.mk#L57-L83

Thus only when this environment variable is non-empty the RISC-V targets defined there are built.

I can't really tell what's happening on your local machine based on only the commands, without the directory from which your executing them and without the output. However, generally you wouldn't use the `build_all.sh` script to build a single application -- this mass-builds every example app. You'll want to execute the `make RISCV=1` command (which sets the `RISCV` variable to `1`) inside of an application directory, e.g. `$LIBTOCKC_ROOT/examples/c_hello`. Running `make` top-level only provides access to generic maintenance targets. When you really want to build all apps, you have to make sure that the `RISCV` environment variable is set in the invocation of the `make` command which `build_all.sh` runs:

https://github.com/tock/libtock-c/blob/5350f80f73949121cb98fc1c215455d4cf22b524/examples/build_all.sh#L32

Generally, you should be able to do that with (depending on your shell) something like: `libtock-c/examples $ RISCV=1 ./build_all.sh`.

bradjc, 2021-09-21:
Also, the readme mentions the RISC-V support.

closed 2021-09-21 by bradjc

nicolast0604, 2021-09-23:
Hi 
Thanks for your patience to explain to me as a beginner in this field.
Then, I run make RISCV=1 in the directory of examples/c_hello.
I have the following tbf, 

./build/cortex-m7/cortex-m7.tbf
./build/rv32imc/rv32imc.0x20032080.0x10008000.tbf
./build/rv32imc/rv32imc.0x20030880.0x10008000.tbf
./build/rv32imc/rv32imc.0x41000060.0x42008000.tbf
./build/rv32imc/rv32imc.0x20034080.0x10008000.tbf
./build/rv32imc/rv32imc.0x20030080.0x10005000.tbf
./build/cortex-m4/cortex-m4.tbf
./build/cortex-m0/cortex-m0.tbf
./build/cortex-m3/cortex-m3.tbf
./build/rv32imac/rv32imac.0x40430060.0x80004000.tbf
./build/rv32imac/rv32imac.0x20040060.0x80002800.tbf
./build/rv32imac/rv32imac.0x403B0060.0x3FCC0000.tbf
./build/rv32imac/rv32imac.0x40440060.0x80007000.tbf
./build/rv32i/rv32i.0x00080060.0x40008000.tbf

As my target is to port to openitan with ibex risc-v, I guess I should select. ./build/rv32i/rv32i.0x00080060.0x40008000.tbf

Thanks a lot.