tock/libtock-c #282: RISC-V Build Requires libstdc++/libsupc++

reynoldsbd, 2022-07-06:
Trying to build for RISC-V but I keep getting this error at link time:

```
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lstdc++
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lsupc++
```

In other words, my build environment does not contain copies of libstdc++ or libsupc++ for RISC-V.

I followed the instructions to install newlib using the prebuilt .deb package. Is there perhaps another similar step I can follow to get a copy of these C++ libs?

Also interesting: I removed the flags from [this line](https://github.com/tock/libtock-c/blob/14b0118654ddc09d55b704c825c5d685261b38b1/Configuration.mk#L250-L251) in Configuration.mk and my build was successful. Perhaps these libs aren't needed for a simple C application?

@bradjc it looks like you're heavily involved with RISC-V support. Is this perhaps something you have context on?

bradjc, 2022-07-07:
You're right that libstdc++ and libsupc++ are not needed to compile c apps.

What I don't know is why even though those libraries have been included in the compilation for years now you don't have them. Getting a riscv toolchain is a mess.

You have done the easiest fix. Maybe we should have some way to only include them for c++ app builds.

reynoldsbd, 2022-07-07:
Out of curiosity, why do these libs need to be explicitly listed for riscv but not for other platforms?

Happy to submit a patch. What do you think about gating them behind RISCV_CPP=1?

alistair23, 2022-07-08:
Hey, so this is a little more complicated

Your build probably doesn't need those libs. But the `ble_passive_scanning` example needs `-lstdc++` for example. It's also entirely possible that someone's out of tree app needs those libs. By including them in the default Makefile we can allow a greater range of compatibility.

As for installing a libc. It's best to try and get one from your distro. Distros have slowly been improving their embedded libc support for RISC-V. An Arch distro is your best bet for that type of thing.

It sounds like you are using a Debian based distro though. Debian stable has support for [picolibc](https://packages.debian.org/stable/devel/picolibc-riscv64-unknown-elf) which you should be able to use instead of newlib. It's well maintained and will contain bug and security fixes not found in older newlib packages.

picolibc seems to have made it to [Ubuntu as well](https://packages.ubuntu.com/jammy/picolibc-riscv64-unknown-elf). So hopefully someone with a Debian machine can test with it and update the README as required

reynoldsbd, 2022-07-08:
I'm not having any trouble with my libc installation. I am able to compile C apps just fine using the prebuilt .deb described in the README.

The core issue is that "C++ on RISC-V" seems to be broken for a number of reasons. In fact, if I try to compile `ble_passive_scanning` for RISC-V according to the README, I don't even make it to the linker step; compilation fails on `#include <cstring>`.

Lack of decent C++ support is definitely an ecosystem issue, not a Tock-specific problem. However, the presence of these C++ flags (`-lstdc++ -lsupc++`) in the Tock makefiles is preventing me from compiling plain C apps for RISC-V. This is why I suggested making "C++ on RISC-V" an opt-in feature via RISCV_CPP=1.

Alternatively, do you have any suggestions on where I can get a hold of C++ libraries for RISC-V?

(Aside - I love the idea of using picolibc from my distro instead of using Tock's vendored libc. But seem to be lots of hardcoded paths in Configuration.mk that would need to be reconciled.)

alistair23, 2022-07-10:
Yeah, so it seems like you are having libc issues, specifically with the C++ libraries. You would expect to get that from newlib or picolib in this case. You might need some different arguments when you build newlib

> Alternatively, do you have any suggestions on where I can get a hold of C++ libraries for RISC-V?

I gave some suggestions in my previous comment.

> (Aside - I love the idea of using picolibc from my distro instead of using Tock's vendored libc. But seem to be lots of hardcoded paths in Configuration.mk that would need to be reconciled.)

Ah, those hardcoded paths are a pain. A simple revert of 00037a844c64ee2b0c4f664aff704a0a4cc4cb30 should be what you want though

reynoldsbd, 2022-07-11:
> You would expect to get that from newlib or picolib...

Been digging a little more. libstdc++ is a component of GCC itself, not newlib/picolib. It is GNU's implementation of the C++ standard library.

https://gcc.gnu.org/onlinedocs/libstdc++/faq.html

My distro (Debian Bullseye) delivers libstdc++ as a separate package from the core GCC toolchain. But as far as I can tell, a packaged version is not available for `riscv64-unknown-elf`.

I can try building my own version locally. If that works, maybe it should be added to the README as a requirement for RISC-V?

alistair23, 2022-07-11:
Good catch! I didn't bother to check if the library was actually in picolib/newlib!

If you build it locally and get it to work a README update would be great

alexandruradovici, 2022-07-12:
I am having the same issue when building on the mac (M1). Interestingly the error is only popping up when building for the rv32i platform. Commenting out https://github.com/tock/libtock-c/blob/14b0118654ddc09d55b704c825c5d685261b38b1/Configuration.mk#L80 solves the problem.

lschuermann, 2022-07-12:
> Interestingly the error is only popping up when building for the rv32i platform.

This is an interesting obversation, given that `rv32i` really should not be of any influence w.r.t. the C and C++ libraries used. One key difference with `rv32i` is that it does not have multiplication support, which might potentially result in linker errors complaining about linking to `muldiv.o` from `libgcc`, but nothing else (see NixOS/nixpkgs#178140). I don't have an M1 Mac but would like to see `rv32i` working (it's a popular target for smaller FPGAs). Perhaps we can debug this over a call @alexandruradovici?

alexandruradovici, 2022-07-12:
> This is an interesting obversation, given that `rv32i` really should not be of any influence w.r.t. the C and C++ libraries used. One key difference with `rv32i` is that it does not have [...]

Yes, sure. When are you available?

lschuermann, 2022-07-13:
Okay, @alexandruradovici and I debugged this today and it appears that the prebuilt RISC-V toolchain Alex and possibly others are using (if I recall correctly, this was provided by SiFive?) does not support the `rv32imc` target. This is unfortunate, as `rv32i`, being a minimal target, is somewhat popular amongst smaller FPGA designs (think FOMU), as well as an interesting target when experimenting with softcores or even writing your own. However, supporting such a target by default makes the `libtock-c` build experience miserable, given the prevalence of toolchains which do not have prebuilt support for it.

Fortunately, tock/tock#2860 actually changed the LiteX simulation target to use a `rv32imc`-enabled VexRiscv softcore CPU. Hence, we can "fix" this issue for now by changing this particular target to use `rv32imc` instead, and everything will work just fine. That said, this "fix" does feel more like a workaround and (again) exposes some major usability issues in the RISC-V toolchain ecosystem (for now). It means that people wanting to build for `rv32i`, for some exotic hardware, small softcore, or to experimen/debug will be on their own for now, given we'd be explicitly dropping upstream support for such a CPU.

commit f5e16436 (2022-07-22): Merge #285

285: Configuration.mk: change litex_sim target definition to rv32imc r=ppannuto a=lschuermann

This is in response to issue tock/libtock-c#282, which exposed that `rv32i` support is not prevalent among prebuilt RISC-V embedded toolchains. Users of such toolchains are faced with linker errors when trying to build a `rv32i` target, which is included by default in all RISC-V builds. As of tock/tock#2860 the `litex_sim` board is actually using a VexRiscv CPU with multiplication and compressed instruction support and hence this target definition can be changed to target `-march=rv32imc` [... cut]
PR #285 (2022-07-22): Configuration.mk: change litex_sim target definition to rv32imc
This is in response to issue tock/libtock-c#282, which exposed that `rv32i` support is not prevalent among prebuilt RISC-V embedded toolchains. Users of such toolchains are faced with linker errors when trying to build a `rv32i` target, which is included by default in all RISC-V builds. As of tock/tock#2860 the `litex_sim` board is actually using a VexRiscv CPU with multiplication and compressed instruction support and hence this target definition can be changed to target `-march=rv32imc` instead. Yet, it does not solve the underl [... cut]
closed 2022-07-22 by bors[bot]