tock/libtock-c #301: Can't build examples with _FORTIFY_SOURCE defined

gemarcano, 2022-09-12:
I am using my own cross compiler (built using Gentoo's `crossdev` utility). ARM GCC is version 12.2.0, an experimental version of 13 for RISCV, and newlib is version 4.2.0.20211231.

When `crossdev` builds GCC, it uses the Gentoo-specific patches for GCC, among which it sets `_FORTIFY_SOURCE=2` when optimizations are enabled. If I leave `_FORTIFY_SOURCE` alone (so, defaulting to defined with a value of 2), I get the following error (a few times):

```
./../../support/warning_header.h:27:5: error: expected declaration specifiers or ‘...’ before numeric constant
   27 | int sprintf(char * restrict str, const char * restrict format, ...);
      |     ^~~~~~~
```

From what I can tell, the problem is that those functions aren't really functions when `_FORTIFY_SOURCE` is not zero. An example from newlib's `ssp/stdio.h`:
```
#define sprintf(str, ...) \
    __builtin___sprintf_chk(str, 0, __ssp_bos(str), __VA_ARGS__)
```

In other words, the declarations in `warning_header.h` are being expanded, and then blow up the compilation.

If I explicitly tell GCC `-U_FORTIFY_SOURCE`, I can compile the examples just fine, as it doesn't include the `ssp/stdio.h` header.

Here's the full log of trying to build `c_hello` with the current `warning_header.h` with `_FORTIFY_SOURCE=2`:
```
$ make V=1 RISCV=1

**************************************************
TOCK USERLAND BUILD SYSTEM -- VERBOSE BUILD
**************************************************
Config:
GIT: release-1.3-768-gad372e5
arm-none-eabi-gcc --version: arm-none-eabi-gcc (Gentoo 12.2.0 p1) 12.2.0 Copyright (C) 2022 Free Software Foundation, Inc. This is free software; see the source for copying conditions.  There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
riscv64-unknown-elf-gcc --version: riscv64-unknown-elf-gcc (Gentoo 13.0.0_pre20220904 p1) 13.0.0 20220904 (experimental) Copyright (C) 2022 Free Software Foundation, Inc. This is free software; see the source for copying conditions.  There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
LAYOUT=../../userland_generic.ld
MAKEFLAGS=-r -R
PACKAGE_NAME=c_hello
TOCK_ARCHS=cortex-m0 cortex-m3 cortex-m4 cortex-m7 rv32imac rv32imc
TOCK_TARGETS=cortex-m0 cortex-m3 cortex-m4 cortex-m7 rv32imac|rv32imac.0x20040060.0x80002800|0x20040060|0x80002800 rv32imac|rv32imac.0x403B0060.0x3FCC0000|0x403B0060|0x3FCC0000 rv32imc|rv32imc.0x41000060.0x42008000|0x41000060|0x42008000 rv32imc|rv32imc.0x00080060.0x40008000|0x00080060|0x40008000 rv32imc|rv32imc.0x20030080.0x10005000|0x20030080|0x10005000 rv32imc|rv32imc.0x20030880.0x10008000|0x20030880|0x10008000 rv32imc|rv32imc.0x20032080.0x10008000|0x20032080|0x10008000 rv32imc|rv32imc.0x20034080.0x10008000|0x20034080|0x10008000 rv32imac|rv32imac.0x40430060.0x80004000|0x40430060|0x80004000 rv32imac|rv32imac.0x40440060.0x80007000|0x40440060|0x80007000
TOCK_USERLAND_BASE_DIR=../..
TOOLCHAIN=
**************************************************

mkdir -p ../../libtock/build/cortex-m0
arm-none-eabi-gcc -std=gnu11 -Wbad-function-cast  -Wmissing-prototypes  -Wnested-externs  -Wold-style-definition  -Wjump-misses-init  -frecord-gcc-switches -gdwarf-2 -Os -fdata-sections -ffunction-sections -fstack-usage -Wall -Wextra -include ../../support/warning_header.h -Wdate-time  -Wfloat-equal  -Wformat-nonliteral  -Wformat-security  -Wformat-y2k  -Winit-self  -Wmissing-declarations  -Wmissing-field-initializers  -Wmissing-format-attribute  -Wmissing-noreturn  -Wmultichar  -Wpointer-arith  -Wredundant-decls  -Wshadow  -Wunused-macros  -Wunused-parameter  -Wwrite-strings  -I../../libtock/ -I../../libtock//include -I../../libtock//../include -I../../libtock/internal/ -I../../libtock/internal//include -I../../libtock/internal//../include -Wstack-usage=2048 -Wlogical-op  -Wtrampolines  -Wl,--emit-relocs -fPIC -mthumb -mfloat-abi=soft -msingle-pic-base -mpic-register=r9 -mno-pic-data-is-text-relative -mcpu=cortex-m0 -march=armv6s-m -MF"../../libtock/build/cortex-m0/alarm_internal.d" -MG -MM -MP -MT"../../libtock/build/cortex-m0/alarm_internal.d@" -MT"../../libtock/build/cortex-m0/alarm_internal.o" "../../libtock/internal/alarm_internal.c"
arm-none-eabi-gcc -std=gnu11 -Wbad-function-cast  -Wmissing-prototypes  -Wnested-externs  -Wold-style-definition  -Wjump-misses-init  -frecord-gcc-switches -gdwarf-2 -Os -fdata-sections -ffunction-sections -fstack-usage -Wall -Wextra -include ../../support/warning_header.h -Wdate-time  -Wfloat-equal  -Wformat-nonliteral  -Wformat-security  -Wformat-y2k  -Winit-self  -Wmissing-declarations  -Wmissing-field-initializers  -Wmissing-format-attribute  -Wmissing-noreturn  -Wmultichar  -Wpointer-arith  -Wredundant-decls  -Wshadow  -Wunused-macros  -Wunused-parameter  -Wwrite-strings  -I../../libtock/ -I../../libtock//include -I../../libtock//../include -I../../libtock/internal/ -I../../libtock/internal//include -I../../libtock/internal//../include -Wstack-usage=2048 -Wlogical-op  -Wtrampolines  -Wl,--emit-relocs -fPIC -mthumb -mfloat-abi=soft -msingle-pic-base -mpic-register=r9 -mno-pic-data-is-text-relative -mcpu=cortex-m0 -march=armv6s-m -c -o ../../libtock/build/cortex-m0/alarm_internal.o ../../libtock/internal/alarm_internal.c
In file included from /usr/arm-none-eabi/include/stdio.h:800,
                 from ./../../support/warning_header.h:6,
                 from <command-line>:
./../../support/warning_header.h:27:5: error: expected declaration specifiers or ‘...’ before numeric constant
   27 | int sprintf(char * restrict str, const char * restrict format, ...);
      |     ^~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/stdio.h:34:
./../../support/warning_header.h:27:5: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
   27 | int sprintf(char * restrict str, const char * restrict format, ...);
      |     ^~~~~~~
./../../support/warning_header.h:30:5: error: expected declaration specifiers or ‘...’ before numeric constant
   30 | int vsprintf(char * restrict str, const char * restrict format, va_list ap);
      |     ^~~~~~~~
./../../support/warning_header.h:30:5: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
   30 | int vsprintf(char * restrict str, const char * restrict format, va_list ap);
      |     ^~~~~~~~
make: *** [../../TockLibrary.mk:154: ../../libtock/build/cortex-m0/alarm_internal.o] Error 1
```

bradjc, 2022-09-14:
I'm not sure how to help with this. What is the intended behavior? Are we doing something with libtock-c which is unusual, or is `FORTIFY_SOURCE` experimental to the point where this type of behavior is uncommon? Is the idea we would build with `FORTIFY_SOURCE` by default in the future?

gemarcano, 2022-09-14:
As far as I can tell, `_FORTIFY_SOURCE` is a define that GLIBC (and apparently Newlib at some later date) added sometime around 2014 that adds buffer overflow checking to a bunch of standard library calls (earliest blog post I could find about it from RedHat: https://www.redhat.com/en/blog/enhance-application-security-fortifysource).

I'm not sure who's not following the standards, but at the very least, at an implementation level Newlib implements the checks for `_FORTIFY_SOURCE` by changing the functions to macros, which clashes with the `warning_header.h` declarations.

I'm not sure what the best approach is to deal with this issue while retaining the deprecation warnings, especially if newlib is the primary libc used for making tock applications with libtock-c. About the best I can think of is using some like:
```
#if !defined(_NEWLIB_VERSION) || !defined(_FORTIFY_SOURCE) || (_FORTIFY_SOURCE < 1)
// declarations with deprecation warnings
#endif
```

This way the warnings remain only if _FORTIFY_SOURCE is not defined or 0, and even if it is it might work on other libc's if there's ever a switch to one. This is still somewhat fragile, as it depends on newlib headers being included prior to the check.

bradjc, 2022-09-15:
I don't think I still entirely follow, but as long as things work on mac and ubuntu, and CI, that is good enough for me.

commit c77e6606 (2022-09-16): replace warning_header.h with _FORTIFY_SOURCE

Fortify source basically does the same thing, only with static checks
of correctness (or runtime injection of checks) on more complete suite
of functions and with more internal compiler support than we can pull
off. Let's use the tool built for this job.

There were a bunch of things that didn't specify headers correctly
and got away with it becuase our build system was injecting the headers
implicitly. That was an unfortunate side effect of our approach...
This fixes that as well.

Closes #301.
commit 8838d5bd (2022-09-16): replace warning_header.h with _FORTIFY_SOURCE

Fortify source basically does the same thing, only with static checks
of correctness (or runtime injection of checks) on more complete suite
of functions and with more internal compiler support than we can pull
off. Let's use the tool built for this job.

There were a bunch of things that didn't specify headers correctly
and got away with it becuase our build system was injecting the headers
implicitly. That was an unfortunate side effect of our approach...
This fixes that as well.

Closes #301.
commit 872f3ee4 (2022-09-21): Merge #303

303: replace warning_header.h with _FORTIFY_SOURCE r=bradjc a=ppannuto

Fortify source basically does the same thing, only with static checks of correctness (or runtime injection of checks) on more complete suite of functions and with more internal compiler support than we can pull off. Let's use the tool built for this job.

There were a bunch of things that didn't specify headers correctly and got away with it becuase our build system was injecting the headers implicitly. That was an unfortunate side effect of our approach... This fixes that as well.

Closes #301.
PR #303 (2022-09-21): replace warning_header.h with _FORTIFY_SOURCE
Fortify source basically does the same thing, only with static checks of correctness (or runtime injection of checks) on more complete suite of functions and with more internal compiler support than we can pull off. Let's use the tool built for this job.

There were a bunch of things that didn't specify headers correctly and got away with it becuase our build system was injecting the headers implicitly. That was an unfortunate side effect of our approach... This fixes that as well.

Closes #301.
closed 2022-09-21 by bors[bot]