tock/libtock-c #304: Cannot build using riscv gcc on Debian

mateibarbu19, 2022-09-23:
Hello, I tried to build some examples for a RISC-V architecture on a Debian machine having installed the requirements specified in this repo's description, but ran into a error. However, using a RISC-V compiler from SiFive's [GNU Embedded Toolchain — v2020.12.8](https://www.sifive.com/software) works.

```bash
$ dpkg -l | grep gcc-riscv64-unknown-elf 
ii  gcc-riscv64-unknown-elf                       8.3.0.2019.08+dfsg-1                   amd64        GCC compiler for embedded RISC-V chips
$ dpkg -l | grep picolibc-riscv64-unknown-elf
ii  picolibc-riscv64-unknown-elf                  1.5.1-2                                all          Smaller embedded C library for RISC-V development
$ uname -a
Linux matei-debian 5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux
$ which riscv64-unknown-elf-gcc
/usr/bin/riscv64-unknown-elf-gcc
$ make RISCV=1
 DIR        ../../libtock/build/cortex-m0
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/internal/nonvolatile_storage_internal.c
  CC        ../../libtock/adc.c
  ...........................................................................................
  AR        ../../libtock/build/cortex-m7/libtock.a
 DIR        ../../libtock/build/rv32imac
  CC        ../../libtock/internal/alarm_internal.c
In file included from ../../libtock/tock.h:4,
                 from ../../libtock/internal/alarm.h:3,
                 from ../../libtock/internal/alarm_internal.c:1:
/usr/lib/gcc/riscv64-unknown-elf/8.3.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                ^~~~~~~~~~
compilation terminated.
make: *** [../../TockLibrary.mk:154: ../../libtock/build/rv32imac/alarm_internal.o] Error 1```

Has anybody experienced this issue before?

ppannuto, 2022-09-23:
`8.3.0.2019.08+dfsg-1` that 2019 is conspicuously old.

I don't believe we have anyone who uses Debian as their primary distribution, so I don't think we have a sense of which packages are necessary on Debian-proper. If you have a path to a working toolchain / install directions for Debian, I imagine we would appreciate a PR updating the README with directions for Debian. @bradjc might have a better sense of the risc-v packaging.

I still live in a pretty ARM-only universe. All I do is keep building RISC-V off by default, because their toolchain installation is sadly still a mess (which, this unfortunately just continues to serve as evidence of :/ ).

alistair23, 2022-09-25:
You are missing a libc, which libtock-c expects.

You can build it yourself, which the README describes, just look for the section `If you would rather compile your own newlib-based libc, follow the steps below. Section [newlib-nano](https://github.com/tock/libtock-c/blob/master/newlib-nano) describes some extra config options to build a size optimised newlib.`

For Debian you should also be able to use picolibc though. You should be able to just install the package https://packages.debian.org/stable/devel/picolibc-riscv64-unknown-elf which includes support for 32-bit RISC-V

alistair23, 2022-09-26:
I had a go at updating the documentation to include instructions using Debian. Try following this and let me know how it goes: https://github.com/alistair23/libtock-c/tree/alistair/rv32-support

mateibarbu19, 2022-09-27:
I've tried the following with no success:
```bash
~/libtock-c/examples/c_hello$ git checkout alistair/rv32-support 
Branch 'alistair/rv32-support' set up to track remote branch 'alistair/rv32-support' from 'origin'.
Switched to a new branch 'alistair/rv32-support'
~/libtock-c/examples/c_hello$ make RISCV=1 PICOLIB=1             
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ieee802154.c
  CC        ../../libtock/ipc.c
  CC        ../../libtock/kv_system.c
  CC        ../../libtock/tock.c
  CC        ../../libtock/udp.c
  CC        ../../libtock/unit_test.c
  AR        ../../libtock/build/cortex-m0/libtock.a
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ieee802154.c
  CC        ../../libtock/ipc.c
  CC        ../../libtock/kv_system.c
  CC        ../../libtock/tock.c
  CC        ../../libtock/udp.c
  CC        ../../libtock/unit_test.c
  AR        ../../libtock/build/cortex-m3/libtock.a
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ieee802154.c
  CC        ../../libtock/ipc.c
  CC        ../../libtock/kv_system.c
  CC        ../../libtock/tock.c
  CC        ../../libtock/udp.c
  CC        ../../libtock/unit_test.c
  AR        ../../libtock/build/cortex-m4/libtock.a
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ieee802154.c
  CC        ../../libtock/ipc.c
  CC        ../../libtock/kv_system.c
  CC        ../../libtock/tock.c
  CC        ../../libtock/udp.c
  CC        ../../libtock/unit_test.c
  AR        ../../libtock/build/cortex-m7/libtock.a
 DIR        ../../libtock/build/rv32imac
  CC        ../../libtock/internal/alarm_internal.c
In file included from <command-line>:31:
./../../support/warning_header.h:6:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.
make: *** [../../TockLibrary.mk:149: ../../libtock/build/rv32imac/alarm_internal.o] Error 1

bradjc, 2022-09-27:
Did you run `sudo apt install picolibc-riscv64-unknown-elf`?

mateibarbu19, 2022-09-27:
As I've already mentioned, yes.
```bash
$ dpkg -l | grep picolibc-riscv64-unknown-elf
ii  picolibc-riscv64-unknown-elf                  1.5.1-2
```

alistair23, 2022-09-27:
For debugging it's useful to see the command being run, that can be enabled with `V=1`.

Also, the README in that branch is apparently wrong, you need to use `PICOLIBC=1` (I have updated the README)

So the entire command should be:

```shell
make PICOLIBC=1 V=1
```

alistair23, 2022-09-28:
https://github.com/tock/libtock-c/pull/305

mateibarbu19, 2022-09-29:
Using your branch and PICOLIB**C** works.

```bash
$ make PICOLIBC=1
 DIR        ../../libtock/build/cortex-m0
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/internal/nonvolatile_storage_internal.c
  CC        ../../libtock/adc.c
  CC        ../../libtock/aes.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ambient_light.c
  CC        ../../libtock/analog_comparator.c
  CC        ../../libtock/app_state.c
  CC        ../../libtock/ble.c
  CC        ../../libtock/button.c
  CC        ../../libtock/buzzer.c
  CC        ../../libtock/console.c
  CC        ../../libtock/crc.c
  CC        ../../libtock/crt0.c
  CC        ../../libtock/dac.c
  CC        ../../libtock/gpio_async.c
  CC        ../../libtock/gpio.c
  CC        ../../libtock/hmac.c
  CC        ../../libtock/humidity.c
  CC        ../../libtock/i2c_master.c
  CC        ../../libtock/i2c_master_slave.c
  CC        ../../libtock/ieee802154.c
  CC        ../../libtock/ipc.c
  CC        ../../libtock/kv_system.c
  CC        ../../libtock/l3gd20.c
  CC        ../../libtock/led.c
  CC        ../../libtock/lps25hb.c
  CC        ../../libtock/lsm303dlhc.c
  CC        ../../libtock/ltc294x.c
  CC        ../../libtock/max17205.c
  CC        ../../libtock/ninedof.c
  CC        ../../libtock/nrf51_serialization.c
  CC        ../../libtock/pca9544a.c
  CC        ../../libtock/proximity.c
  CC        ../../libtock/read_only_state.c
  CC        ../../libtock/rng.c
  CC        ../../libtock/screen.c
  CC        ../../libtock/sdcard.c
  CC        ../../libtock/sha.c
  CC        ../../libtock/sound_pressure.c
  CC        ../../libtock/spi.c
  CC        ../../libtock/spi_peripheral.c
  CC        ../../libtock/sys.c
  CC        ../../libtock/temperature.c
  CC        ../../libtock/text_screen.c
  CC        ../../libtock/tock.c
  CC        ../../libtock/touch.c
  CC        ../../libtock/tsl2561.c
  CC        ../../libtock/udp.c
  CC        ../../libtock/unit_test.c
  CC        ../../libtock/usb.c
  AR        ../../libtock/build/cortex-m0/libtock.a
 DIR        ../../libtock/build/cortex-m3
  CC        ../../libtock/internal/alarm_internal.c
  CC        ../../libtock/internal/nonvolatile_storage_internal.c
  CC        ../../libtock/adc.c
  CC        ../../libtock/aes.c
  CC        ../../libtock/alarm_timer.c
  CC        ../../libtock/ambient_light.c
  CC        ../../libtock/analog_comparator.c
  CC        ../../libtock/app_state.c
  CC        ../../libtock/ble.c
  CC        ../../libtock/button.c
  CC        ../../libtock/buzzer.c
  CC        ../../libtock/console.c
[... 276 lines cut ...]
  LD        build/rv32imac/rv32imac.0x40430060.0x80004000.elf
  LD        build/rv32imac/rv32imac.0x40440060.0x80007000.elf
 E2T        build/c_hello.tab
Application size report for arch family cortex-m:
   text    data     bss     dec     hex filename
    836     208    3420    4464    1170 build/cortex-m0/cortex-m0.elf
    840     208    3420    4468    1174 build/cortex-m3/cortex-m3.elf
    840     208    3420    4468    1174 build/cortex-m4/cortex-m4.elf
    840     208    3420    4468    1174 build/cortex-m7/cortex-m7.elf
   3356     832   13680   17868    45cc (TOTALS)
Application size report for arch family rv32i:
   text    data     bss     dec     hex filename
   2212      88    3464    5764    1684 build/rv32imac/rv32imac.0x20040060.0x80002800.elf
   2212      88    3464    5764    1684 build/rv32imac/rv32imac.0x403B0060.0x3FCC0000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x41000060.0x42008000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x00080060.0x40008000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x20030080.0x10005000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x20030880.0x10008000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x20032080.0x10008000.elf
   2548      88    3464    6100    17d4 build/rv32imc/rv32imc.0x20034080.0x10008000.elf
   2212      88    3464    5764    1684 build/rv32imac/rv32imac.0x40430060.0x80004000.elf
   2212      88    3464    5764    1684 build/rv32imac/rv32imac.0x40440060.0x80007000.elf
  24136     880   34640   59656    e908 (TOTALS)

```

elderbig, 2022-10-10:
the same error before,after build & install newlib-4.1.0.tar.gz on ubuntu 22,got new error:

```
cd c_hello
make RISCV=1
......
....
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lstdc++
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lsupc++
collect2: error: ld returned 1 exit status
make: *** [../../AppMakefile.mk:269：build/rv32imac/rv32imac.0x20040060.0x80002800.elf] 错误 1

```

alistair23, 2022-10-10:
Can you try following the instructions in https://github.com/tock/libtock-c/pull/305

bradjc, 2024-02-09:
With the new build system and our own newlib this shouldn't be an issue.

closed 2024-02-09 by bradjc