tock/libtock-c #307: What happened to alarm_read?

reynoldsbd, 2022-10-05:
_alarm.h_ declares a function named `alarm_read`. Ostensibly this is used to read the current counter value of the underlying alarm hardware using [command number 2 of the alarm driver](https://github.com/tock/tock/blob/master/doc/syscalls/00000_alarm.md).

Unfortunately this function doesn't seem to be defined anywhere. GitHub search suggests it _used_ to be in a file called _alarm.c_, but that file doesn't exist today (and having trouble tracking down the commit where it was removed).

reynoldsbd, 2022-10-05:
Thanks for the background!

The app I'm working on requires a steady source of ticks, and the alarm driver feels like the right fit. It's easy enough to issue command 2 manually, but all the better if the logic can come as part of libtock-c

brghena, 2022-10-05:
I think it is a bug though that the function is straight-up missing. An implementation almost certainly just calls `alarm_internal_read()`, but we should make it exist. Probably added to `alarm_timer.c`.

bradjc, 2023-07-26:
The empty function declaration removed in #310