tock/libtock-c #352: Issues building `libtock-c` with Nix

lschuermann, 2023-10-31:
Originally posted by @Samir-Rashid on tock/libtock-c#349

I got this error on some examples when running `examples/build_all.sh` in a `nix-shell`.
> can't link soft-float modules with double-float modules

I went spelunking in issues which ended up here https://github.com/tock/tock/issues/1246. I was able to do a workaround (sources [1](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/836), [2](https://stackoverflow.com/questions/65036779/cant-link-double-float-modules-with-soft-float-modules-riscv-compiler)). This is not a permanent fix, but I monkey patched `--enable-multilib` to get things to work locally. I see there is some stuff noted in the readme, but the `build_all` script should just work with no intervention.

<details><summary>Diff with everything I did to get things to work</summary>
<p>

```diff
diff --git a/Configuration.mk b/Configuration.mk
index 4e71e02..0f6b327 100644
--- a/Configuration.mk
+++ b/Configuration.mk
@@ -129,7 +129,7 @@ ifndef ELF2TAB_EXISTS
   ELF2TAB_VERSION := $(shell $(SHELL) -c "$(ELF2TAB) --version | cut -d ' ' -f 2")
   UPGRADE_ELF2TAB := $(shell $(SHELL) -c "printf '%s\n%s\n' '$(ELF2TAB_REQUIRED_VERSION)' '$(ELF2TAB_VERSION)' | sort --check=quiet --version-sort || echo yes")
   ifeq ($(UPGRADE_ELF2TAB),yes)
-    $(error Failed to automatically update elf2tab, please update manually elf2tab to >= $(ELF2TAB_REQUIRED_VERSION))
+#     $(error Failed to automatically update elf2tab, please update manually elf2tab to >= $(ELF2TAB_REQUIRED_VERSION))
   endif
 endif
 
diff --git a/shell.nix b/shell.nix
index 640361c..da78c32 100644
--- a/shell.nix
+++ b/shell.nix
@@ -54,6 +54,7 @@ in
       gcc-arm-embedded
       python3Full
       tockloader
+      uncrustify
     ] ++ (lib.optionals withUnfreePkgs [
       segger-jlink
       tockloader.nrf-command-line-tools
@@ -70,7 +71,7 @@ in
             "--without-headers"
             "--disable-shared"
             "--disable-libssp"
-            "--disable-multilib"
+            "--enable-multilib"
             "--with-newlib"
           ];
           gcc_cv_libc_provides_ssp = "yes";
diff --git a/tools/uncrustify/uncrustify.sh b/tools/uncrustify/uncrustify.sh
index 285e0fd..6f69665 100755
--- a/tools/uncrustify/uncrustify.sh
+++ b/tools/uncrustify/uncrustify.sh
@@ -17,9 +17,7 @@ if ! command -v uncrustify >/dev/null; then
 else
   # Validate uncrustify version
   VERSION=$(uncrustify --version | egrep -o '0.[0-9]+[.0-9]*')
-  if [[ "$VERSION" != $UNCRUSTIFY_VERSION ]]; then
-    do_install=true
-  fi
+
 fi
 
 # install uncrustify if it's missing

```

</p>
</details> 

With this, everything builds except for `./tests/mpu_stack_growth` which I saw is a [problematic test](https://github.com/tock/libtock-c/issues/290), so I did not debug it.
The error itself is just an infinite recursion warning.
<details><summary>Error output</summary>
<p>

```C
main.c: In function 'grow_stack':
main.c:26:13: error: infinite recursion detected [-Werror=infinite-recursion]
   26 | static void grow_stack(void) {
      |             ^~~~~~~~~~
```

</p>
</details> 

### Other notes

Why do none of the Nix user Tock maintainers post their nix configs? It took me hours to scour scant NixOS documentation to see I need to set this to fix udev issues.
```nix
# enable udev rules from packages
services.udev.packages = [
      pkgs.segger-jlink
];
```
Similarly for this libtock-c `shell.nix`, I had to further edit it to allow unfree and accept jlink EULA. Why can't these lines be included and commented out (like [this](https://github.com/tock/tock/pull/3521/files) but commented)?

_Originally posted by @Samir-Rashid in https://github.com/tock/libtock-c/issues/349#issuecomment-1783741217_

lschuermann, 2023-10-31:
> ```diff
> +      uncrustify
> ```

Orthogonal change, but looks good to me. You'll want to revert the uncommenting of the `elf2tab` and `uncrustify` update mechanisms. They don't work on NixOS but may work on other distributions. I have a snippet which prompts the user whether the Makefile should attempt to install `elf2tab` before doing so, which I can try to dig up from my pile of git branches.

> I got this error on some examples when running `examples/build_all.sh` in a `nix-shell`.
> 
> > can't link soft-float modules with double-float modules

tock/tock#1246 concerns ARM Cortex-M only, not RISC-V. The changes you're proposing only affect the custom-built RISC-V toolchain which `shell.nix` defines in this repo. I spent weeks coming up with those compiler flags, and apart from an issue with `rv32i` (no extensions) RISC-V targets, they seemed to work file for a long time. This led to a bunch of discussions with upstream Nixpkgs maintainers, both offline and here: https://github.com/NixOS/nixpkgs/pull/178140. Truth be told, there was a lot of guesswork involved in coming up with the newlib configuration flags, so it is possible that `enable-multilib` is required. If things still compile, I'll happily accept those changes.

However, hopefully https://github.com/tock/libtock-c/pull/346 will alleviate the need to compile a custom toolchain entirely. IIRC we just have to sort out some licensing & artifact distribution questions there.

> Why do none of the Nix user Tock maintainers post their nix configs?

I cannot speak for others, but I've leaked secrets in my once-public config more often than I'd like to admit. I once had the goal of having a CI-checked NixOS VM with all toolchains pre-installed, but that's (a) work and (b) pretty exotic. I built a HashiCorp Packer VM for the last Tock tutorial, but I'm unsure how well this continues to work in the future. Building this was painful.

> Similarly for this libtock-c `shell.nix`, I had to further edit it to allow unfree and accept jlink EULA. Why can't these lines be included and commented out (like [this](https://github.com/tock/tock/pull/3521/files) but commented)?

Reiterating my comment from https://github.com/tock/tock/pull/3727: J-Link is only used by a subset of our boards, and generally also requires installation steps on other platforms. I'm unsure whether we should and can ship instructions for installing the (proprietary) toolchains for every board on every OS. What we most certainly don't want to do is vendor unfree packages / accept EULAs etc. for the user. That being said, adding comments or OS-specific install instructions somewhere can't hurt for sure!

Samir-Rashid, 2024-01-26:
This issue is stale. Please close.

lschuermann, 2024-01-26:
@Samir-Rashid Correct, thanks!

closed 2024-01-26 by lschuermann