tock/libtock-c #372: Making static memory allocation optional

gentooza, 2024-03-14:
Hi!

Is there any option to tell GCC to not use PIC and *hardcoding* an application memory layout as done in libtock-rs

I know the interest is to make dynamic memory allocation universal to every language and architecture, but I think it could be interesting for testing purposes.

Cheers

ppannuto, 2024-03-20:
All the RISC-V platforms currently do that [as there isn't yet PIC support for RISC-V]. If you're testing some kind of automated tool, probably easiest to just use a RISC-V build. If you need ARM, you can hopefully use the configuration from RISC-V to get something working locally, but I don't think it is something we would be interested in maintaining as a build configuration option upstream—the build system is pretty complex as-is, and relocatable apps are pretty central to Tock's design/philosophy.

bradjc, 2024-03-22:
I think it should be possible to set `TOCK_TARGET` with something like `cortex-m4|cortexm4.0x20040080.0x80002800|0x20040080|0x80002800`, just like RISC-V. Not tested, but I don't recall any special checks for rv32 platforms.

gentooza, 2024-04-01:
Cool, perfect!

I'll check the compiling process for Risc-V and test it in ARM.

Thanks

ppannuto, 2024-04-02:
I'm going to close this as resolved for now. Feel free to re-open if there's still issues or questions here.

closed 2024-04-02 by ppannuto