tock/libtock-c #377: remove mention of RISCV flag support in readme

lauralt, 2024-03-22:
Hi,

I've been trying to build the `c_hello` application for a Cortex-M target with WSL. I've noticed in the Readme that there is a RISCV flag that can enable/disable the compilation for the riscv targets. From what it looks like from the scripts, RISCV flag is no longer supported. The README should be updated accordingly in that case.

Also, is there a way to compile an application only on Cortex-M (without commenting the riscv targets in Configuration.mk :D)? If not, why did you choose to build on both of them as the default, and I think, the only option.

lauralt, 2024-03-25:
It worked, thanks! I'll submit a PR in the following days.

lauralt, 2024-03-26:
@lschuermann  I see the flag was removed in https://github.com/tock/libtock-c/commit/4de0e701932a3ff8565681e4dd172055ce78d9f5, so I will close the issue.

Might be worth documenting also the `TOCK_TARGETS` option you mentioned in the README.

closed 2024-03-26 by lauralt

eddyp, 2024-03-27:
> @lschuermann I see the flag was removed in [4de0e70](https://github.com/tock/libtock-c/commit/4de0e701932a3ff8565681e4dd172055ce78d9f5), so I will close the issue.
> 
> Might be worth documenting also the `TOCK_TARGETS` option you mentioned in the README.

I agree documenting `TOCK_TARGETS` especially since one might be interested in an Arm Cortex M target, but being forced to also install a RISC-V toolchain makes little sense from that PoV. Also, for build speed reasons it makes sense to be able to selectively choose what is to be compiled and ran.